ARMmbed/mbed-drivers #98: PinNames replaced by GeneratedPins

0xc0170, 2015-10-02:
This is the top tracking item, as this PinNames are used in this layer and it should propagate to lower layers - hal and targets. As soon as drivers version 0.10.0 comes, there will be new HAL which defines GeneratedPins, which should enable us to switch to this header and remove all legacy-header PinNames inclusion

bremoran, 2015-10-02:
Please don't do this!

bremoran, 2015-10-02:
PinNames.h has historically been the include that is used for pin names.  I do not intend this to change.  The process going forward should be this:

1. Add yotta config pin names to targets
1. as pin names are added to targets, place a PinNames.h in ```mbed-hal-<vendor>-<chip>```
1. Remove PinNames.h from ```mbed-hal-<board>```
1. Once all targets have yotta pin configuration, place a PinNames.h in ```mbed-hal```
1. Gradually remove PinNames.h from ```mbed-hal-<vendor>-<chip>```

0xc0170, 2015-10-02:
It is currently confusing how it is, therefore I created this issue. 

> Once all targets have yotta pin configuration, place a PinNames.h in mbed-hal

Meaning that there will be generated pinname header file, and then PinNames in mbed-hal?

> Gradually remove PinNames.h from mbed-hal-<vendor>-<chip> 

It would be possibly to do clean this within a specific version mbed-hal? As we have now limited number of targets

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-842

rainierwolfcastle, 2015-10-16:
ARM Internal Ref: IOTSFW-1011

closed 2015-10-26 by 0xc0170