ARMmbed/mbed-drivers #103: Version mismatch error in latest build

tabarr, 2015-10-19:
Did a yt update and yt build on my copy of mbed-client-example-6lowpan project. Now getting some sort of version mismatch error with mbed-drivers. I can run yt update multiple times and get the same download message for the 5 modules as show below. 

(workspace) C:\mbedos\projects\mbed-client-example-6lowpan>yt target
frdm-k64f-gcc 0.1.4
mbed-gcc 0.1.3

(workspace) C:\mbedos\projects\mbed-client-example-6lowpan>yt update
info: get versions for frdm-k64f-gcc
info: get versions for mbed-gcc
info: get versions for mbed-client
info: get versions for mbed-client-c
info: get versions for mbed-client-mbed-os
info: get versions for mbed-client-mbedtls
info: get versions for nanostack-libservice
info: get versions for mbed-drivers
info: get versions for sockets
info: get versions for mbed-hal
info: download mbed-hal@1.0.0 from the public module registry
info: get versions for cmsis-core
info: download cmsis-core@1.0.0 from the public module registry
info: get versions for ualloc
info: download ualloc@1.0.0 from the public module registry
info: get versions for minar
info: download minar@1.0.0 from the public module registry
info: get versions for core-util
info: download core-util@1.0.0 from the public module registry
info: get versions for compiler-polyfill
info: get versions for mbed-hal-freescale
info: get versions for mbed-hal-ksdk-mcu
info: get versions for uvisor-lib
info: get versions for mbed-hal-k64f
info: get versions for cmsis-core-freescale
info: get versions for cmsis-core-k64f
info: get versions for mbed-hal-frdm-k64f
info: get versions for dlmalloc
info: get versions for minar-platform
info: get versions for minar-platform-mbed
info: get versions for sal
info: get versions for sal-stack-lwip
info: get versions for sal-driver-lwip-k64f-eth
info: get versions for sal-iface-eth
info: get versions for mbedtls

(workspace) C:\mbedos\projects\mbed-client-example-6lowpan>yt build
info: get versions for mbed-hal
info: download mbed-hal@1.0.0 from the public module registry
info: get versions for cmsis-core
info: download cmsis-core@1.0.0 from the public module registry
info: get versions for ualloc
info: download ualloc@1.0.0 from the public module registry
info: get versions for minar
info: download minar@1.0.0 from the public module registry
info: get versions for core-util
info: download core-util@1.0.0 from the public module registry
error: mbed-hal does not meet specification ^1.0.0 required by mbed-drivers
error: cmsis-core does not meet specification ^1.0.0 required by mbed-drivers
error: ualloc does not meet specification ^1.0.0 required by mbed-drivers
error: minar does not meet specification ^1.0.0 required by mbed-drivers
error: core-util does not meet specification ^1.0.0 required by mbed-drivers
info: generate for target: frdm-k64f-gcc 0.1.4 at C:\mbedos\projects\mbed-client-example-6lowpan\yotta_targets\frdm-k64f
-gcc
GCC version is: 4.9.3
-- Configuring done
-- Generating done
-- Build files have been written to: C:/mbedos/projects/mbed-client-example-6lowpan/build/frdm-k64f-gcc
ninja: no work to do.

autopulated, 2015-10-19:
@tabarr the repeated download messages indicate that something might be corrupt in your node modules folder, so it's worth removing that and trying again.

That said we're currently dealing with some issues resulting from the recent bumping of everything to 1.0. Will update here when those are resolved.

tabarr, 2015-10-19:
I did try what you suggested and was able to get rid of the issues with mbed-drivers. But now I have error messages for other modules. Can I assume these are related to the general issues you are having with bumping modules to 1.0?

error: cmsis-core does not meet specification ~0.4.0 required by mbedtls
error: minar does not meet specification <1.0.0 required by sal-stack-nanostack-eventloop
error: core-util does not meet specification ~0.3.0 required by sockets

bogdanm, 2015-10-19:
Hi,

We are still working on the 1.0.0 update, things might temporary break until we get that done. Sorry for this, we hope we'll be able to fix the remaining issues soon.

Thanks,
Bogdan

simonqhughes, 2015-10-21:
ARM Internal Ref: IOTSFW-1060

0xc0170, 2015-10-26:
The issue should be fixed now

closed 2015-10-26 by 0xc0170