ARMmbed/mbed-drivers #107: Test failures when running mbedgt ok K64F

c1728p9, 2015-10-27:
When running greentea tests on v0.11.1 - df38aa98e51761facfba7e4785014fcbf245c2ec there are multiple IOERR_SERIAL errors along with a failure.  This is using cmsis-dap software version 0226 on a K64F freedom board, with the host pc running windows 7 (on real hardware, not a VM). 

GCC output:
```
+---------------+---------------+------------------------------------+--------------+--------------------+-------------+
| target        | platform_name | test                               | result       | elapsed_time (sec) | copy_method |
+---------------+---------------+------------------------------------+--------------+--------------------+-------------+
| frdm-k64f-gcc | K64F          | mbed-drivers-test-serial_interrupt | OK           | 6.19               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-blinky           | OK           | 4.43               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-div              | IOERR_SERIAL | 16.99              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-dev_null         | OK           | 3.51               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-stdio            | IOERR_SERIAL | 17.19              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-sleep_timeout    | OK           | 3.42               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-ticker           | IOERR_SERIAL | 16.94              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-rtc              | IOERR_SERIAL | 17.69              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-cstring          | FAIL         | 1.41               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-cpp              | OK           | 1.38               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-timeout          | IOERR_SERIAL | 16.92              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-basic            | OK           | 1.35               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-ticker_3         | IOERR_SERIAL | 16.89              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-ticker_2         | OK           | 11.38              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-heap_and_stack   | OK           | 1.39               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-hello            | OK           | 0.38               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-time_us          | OK           | 11.38              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-stl              | OK           | 1.38               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-echo             | IOERR_SERIAL | 17.43              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-detect           | OK           | 0.45               | shell       |
+---------------+---------------+------------------------------------+--------------+--------------------+-------------+

Result: 1 FAIL / 12 OK / 7 IOERR_SERIAL
Completed in 267.58 sec
```

ARMCC output:
```
+-----------------+---------------+------------------------------------+--------------+--------------------+-------------+
| target          | platform_name | test                               | result       | elapsed_time (sec) | copy_method |
+-----------------+---------------+------------------------------------+--------------+--------------------+-------------+
| frdm-k64f-armcc | K64F          | mbed-drivers-test-serial_interrupt | OK           | 6.17               | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-blinky           | OK           | 4.44               | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-div              | IOERR_SERIAL | 16.96              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-dev_null         | OK           | 3.51               | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-stdio            | IOERR_SERIAL | 16.94              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-sleep_timeout    | OK           | 2.4                | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-ticker           | IOERR_SERIAL | 17.14              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-rtc              | IOERR_SERIAL | 17.02              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-cstring          | OK           | 1.93               | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-cpp              | OK           | 1.37               | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-timeout          | IOERR_SERIAL | 16.91              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-basic            | OK           | 1.36               | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-ticker_3         | IOERR_SERIAL | 16.95              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-ticker_2         | OK           | 11.4               | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-heap_and_stack   | OK           | 1.4                | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-hello            | OK           | 0.39               | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-time_us          | OK           | 11.38              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-stl              | OK           | 1.91               | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-echo             | IOERR_SERIAL | 17.13              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-detect           | OK           | 0.46               | shell       |
+-----------------+---------------+------------------------------------+--------------+--------------------+-------------+

Result: 13 OK / 7 IOERR_SERIAL
Completed in 268.98 sec
```

0xc0170, 2015-10-28:
@c1728p9 Can you test any other module? I believe this is an issue for mbed greentea.

cc @PrzemekWirkus

PrzemekWirkus, 2015-10-28:
Hi,
I think ```0226``` works better on Linux, for Windows could you please downgrade to older version (without DETAILS.TXT file on it) and try again?

c1728p9, 2015-10-28:
0203 and 0221 work well on windows.  Both only fail one test and have no IO errors.

```
+---------------+---------------+------------------------------------+--------+--------------------+-------------+
| target        | platform_name | test                               | result |elapsed_time (sec) | copy_method |
+---------------+---------------+------------------------------------+--------+--------------------+-------------+
| frdm-k64f-gcc | K64F          | mbed-drivers-test-serial_interrupt | OK     |6.23               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-blinky           | OK     |4.41               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-div              | OK     |1.36               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-dev_null         | OK     |3.49               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-stdio            | OK     |0.75               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-sleep_timeout    | OK     |2.37               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-ticker           | OK     |11.37              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-rtc              | OK     |4.58               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-cstring          | FAIL   |1.37               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-cpp              | OK     |1.37               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-timeout          | OK     |11.55              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-basic            | OK     |1.35               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-ticker_3         | OK     |11.35              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-ticker_2         | OK     |11.4               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-heap_and_stack   | OK     |1.39               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-hello            | OK     |0.38               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-time_us          | OK     |11.39              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-stl              | OK     |1.38               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-echo             | OK     |6.31               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-detect           | OK     |0.47               | shell       |
+---------------+---------------+------------------------------------+--------+--------------------+-------------+

Result: 1 FAIL / 19 OK
Completed in 248.46 sec
```

PrzemekWirkus, 2015-10-28:
@c1728p9 THat's a good news. I'm in process of investigating which interface chip version is most suitable for Win7 and Ubuntu.

PrzemekWirkus, 2015-11-03:
interface chip for K64F:
Windows: "20140530_k20dx128_k64f_if_mbed.bin"
Ubuntu: "0226_k20dx128_k64f_0x5000.bin"

0xc0170, 2015-11-03:
Please redirect this to mbedgt. @c1728p9 I would like to know which firmware we could use for each OS.

I'll create a new issue for cstring failure for gcc targets.

closed 2015-11-03 by 0xc0170