ARMmbed/mbed-drivers #108: Failure when testing with the st-stm32f439zi-gcc

c1728p9, 2015-10-27:
I get this error when trying to build the st-stm32f439zi-gcc on windows 7:

```
C:/projects/armmbed/mbed-drivers/yotta_modules/mbed-hal-st-stm32f429zi/mbed-hal-st-stm32f429zi/objects.h:97:5: error: 'PinName' does not name a type
     PinName pin;
     ^
In file included from C:/projects/armmbed/mbed-drivers/yotta_modules/mbed-hal-st-stm32f429zi/mbed-hal-st-stm32f429zi/objects.h:108:0,
                 from C:/projects/armmbed/mbed-drivers/yotta_modules/mbed-hal-st-stm32f429zi/mbed-hal-st-stm32f429zi/device.h:71,
                 from C:/projects/armmbed/mbed-drivers/mbed-drivers/platform.h:21,
                 from C:/projects/armmbed/mbed-drivers/mbed-drivers/Stream.h:19,
                 from C:/projects/armmbed/mbed-drivers/source/Stream.cpp:16
:
C:/projects/armmbed/mbed-drivers/yotta_modules/mbed-hal-st-stm32f4/mbed-hal-st-stm32f4/gpio_object.h:44:5: error: 'PinName' does not name a type
     PinName  pin;
     ^
C:/projects/armmbed/mbed-drivers/yotta_modules/mbed-hal-st-stm32f4/mbed-hal-st-stm32f4/gpio_object.h: In function 'int gpio_is_connected(const gpio_t*)':
C:/projects/armmbed/mbed-drivers/yotta_modules/mbed-hal-st-stm32f4/mbed-hal-st-stm32f4/gpio_object.h:67:17: error: 'const struct gpio_t' has no member named 'pin'
     return obj->pin != (PinName)NC;
                 ^
C:/projects/armmbed/mbed-drivers/yotta_modules/mbed-hal-st-stm32f4/mbed-hal-st-stm32f4/gpio_object.h:67:25: error: 'PinName' was not declared in this scope

     return obj->pin != (PinName)NC;
                         ^
C:/projects/armmbed/mbed-drivers/yotta_modules/mbed-hal-st-stm32f4/mbed-hal-st-stm32f4/gpio_object.h:67:33: error: expected ';' before 'NC'
     return obj->pin != (PinName)NC;
                                 ^
C:/projects/armmbed/mbed-drivers/yotta_modules/mbed-hal-st-stm32f4/mbed-hal-st-stm32f4/gpio_object.h:67:33: error: 'NC' was not declared in this scope
[16/193] Building CXX object source/CM...ed/mbed-drivers/source/CallChain.cpp.o
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed
```

winfred-lu, 2015-10-28:
It looks like YOTTA_CFG_HARDWARE_PINS should be defined.
I fixed the issue with adding
    "hardware-pins": {},
to config section of yotta_targets/st-stm32f439zi-gcc/target.json
I am not an export for yotta. Maybe there is some other more proper place to config.

niklas-arm, 2015-10-28:
The `st-stm32f439zi-gcc` target is not fully supported.
Please use the `stm32f429i-disco-gcc` target instead.

@0xc0170 @bremoran

0xc0170, 2015-10-28:
This is not related to mbed-drivers. 

It's a problem of a target, I created a tracking issue in a target repository (currently private repo - https://github.com/ARMmbed/target-st-stm32f439zi-gcc/issues/6). And as noted above, not fully supported.

closed 2015-10-28 by 0xc0170

c1728p9, 2015-10-28:
I'm able to build with stm32f429i-disco-gcc