ARMmbed/mbed-drivers #112: Test cstring failure for gcc targets

0xc0170, 2015-11-03:
The test was ported from mbed, where floats are enabled for libraries, which is not a case for our current gcc targets, thus this test fail for gcc, but not for armcc. We should remove floats testing as it's not true for all targets (issue tracking under mbed-target-gcc: https://github.com/ARMmbed/target-mbed-gcc/issues/17)

File reference: https://github.com/ARMmbed/mbed-drivers/blob/master/test/cstring/main.cpp#L63

bogdanm, 2015-11-04:
I believe this test should be ignored for now, until we figure out how to fix the floating point issue in mbedOS.

rainierwolfcastle, 2015-11-04:
ARM Internal Ref: IOTSFW-1167

PrzemekWirkus, 2015-11-23:
There should be a simple way to disable / ignore test cases -> https://github.com/ARMmbed/greentea/issues/36

autopulated, 2015-11-23:
If you're running tests with `yotta test`, you can add them to `.yotta_ignore` to ignore them.