ARMmbed/mbed-drivers #113: Needs better description and keywords

autopulated, 2015-11-05:
It's hard to find this module in search because of the minimal description and keywords:

https://github.com/ARMmbed/mbed-drivers/blob/master/module.json#L4

(see also https://github.com/ARMmbed/yotta-registry/issues/20)

0xc0170, 2015-11-05:
I saw that error was not certain if its registry or this module. Lets update the description and keywords

autopulated, 2015-11-05:
I'd suggest:
```JSON
{
  "description": "mbed OS drivers, with classes supporting SPI, I2C, UART, GPIO and ADC peripherals. This module is the core of mbed OS, on which most applications depend.",
  "keywords": ["mbed", "mbed-official", "drivers", "spi", "i2c", "uart", "gpio", "adc", "hardware", "abstraction"]
}
```

PrzemekWirkus, 2015-11-05:
Are you willing to call classes in this module ```foundation classes```?

autopulated, 2015-11-05:
That might be a good description, but as we don't use that language anywhere else AFAIK probably isn't sensible to add here.

PR #114 (2015-11-05): improve description and keywords
fixes #113 

@bogdanm @0xc0170
closed 2015-11-05 by autopulated