ARMmbed/mbed-drivers #118: mbed-drivers/app.h should be relocated to core-util

rgrover, 2015-11-18:
Using the header app.h should not require bringing in all of mbed-drivers. In particular, app.h is needed by the POSIX port of Minar, which needs to remain clear of mbed-drivers.

0xc0170, 2015-11-18:
True , declaration should be elsewhere (core-util), definition for mbed will be here.

autopulated, 2015-11-19:
I don't think it necessarily should be moved – using `minar` doesn't imply that you're using app_start. The logic associated with replacing main with app_start is contained in this module, so if you define app_start without including mbed-drivers it won't work.

rgrover, 2015-11-19:
Can we then extract the retarget mechanism into its own module?
POSIX can provide its own retarget.cpp, but it can't afford to pull mbed-drivers.

autopulated, 2015-11-19:
_Should_ non-mbed OS programs use `app_start` at all though? That's the mbed OS entry point.

rgrover, 2015-11-19:
@autopulated are we defining non-mbed-drivers as non-mbed OS?

autopulated, 2015-11-19:
I would think so?

rgrover, 2015-11-19:
MINAR based applications can be run usefully under POSIX, and that doesn't require mbed-drivers. And mbed-drivers need not have a port for all platforms where we wish to run mbed OS.

bogdanm, 2015-11-23:
I tend to agree with @rgrover, mostly because it keeps things simple for us: if we want to run MINAR tests under POSIX, for example, we want them to start with `app_start` instead of `main`, so we can keep the code simple. At this point, I don't see why we'd need to break the `app_start` rule for POSIX.

bremoran, 2015-11-23:
It's trivial to define a main that schedules ```app_start```. That said, the code is effectively common to all minar-based applications, which is the reason for ```app_start``` in the first place. If ```app.h``` is going to move, should it move into minar?

autopulated, 2015-11-23:
if it's just a matter of tests, it should be the test framework that provides a consistent API though, should it not?

On posix you could reasonably want to use minar in a single thread of your application: if MINAR pulls in a library that defines main, that becomes very hard to do.

bogdanm, 2015-11-24:
It's not about a consistent API of the test framework, it's about having the same point (`app_start`) in boht mbed OS and POSIX, as opposed to having `app_start` in mbed OS and `main` in POSIX. I don't think threading is an issue right now.

autopulated, 2015-11-24:
You're thinking about it backwards though: the module that defines app_start **by definition** must implement main.

The question is really: "should `main` be implemented in mbed-drivers, or should `main` be implemented in core-util".

bogdanm, 2015-11-24:
> should main be implemented in mbed-drivers, or should main be implemented in core-util

What about `minar`? Since, after all, the current version of `main` is particular to `minar`.

autopulated, 2015-11-24:
Yes, because mbed OS (mbed-drivers) uses MINAR in a particular way (i.e. in the main thread).

you don't need a function called app_start to use minar, you just need to start the scheduler somewhere.

jrobeson, 2015-12-05:
glib's event loops allows me to use main() as normal, I would expect minar to be the same.

hugovincent, 2015-12-14:
@autopulated @bogdanm I'd have thought putting it in `minar` makes the most sense.

autopulated, 2015-12-14:
@hugovincent OK, but on the understanding that it requires minar to override main, making it difficult to use minar without completely restructuring an existing application (for example it'd be hard to use in a single thread alongside other things in a user-mode program on Linux/ an RTOS).

hugovincent, 2015-12-14:
What about `minar-platform-mbed`?

autopulated, 2015-12-14:
That would be better, but a bit confusing I think – if it's there, why not leave it in mbed-drivers, since that module has a dependency on mbed-drivers?

bogdanm, 2015-12-18:
I vote for moving `app.h` into `minar`, but keeping the implementation of `main` in `mbed-drivers`. For POSIX, we can implement `main` in `minar-platform-posix`, which is actually the case right now.
I want to move this forward (since it is needed for MINAR tests), so if I don't get any strong oppinions against this in the next 2 hours, or if I'll get +1s before that, I'll make it happen.

autopulated, 2015-12-18:
I still think that's the wrong thing to do. app_start is an mbed_os API, not a minar API. You don't need to use app_start to test/use minar.

bogdanm, 2015-12-18:
> You don't need to use app_start to test/use minar.

I'd be totally for that, if all the tests that we have in MINAR wouldn't use `app_start` already, which feels quite natural to me. What would you do instead?

autopulated, 2015-12-18:
Well what's missing is a consistent way of writing tests: something which abstracts app_start (or just provides it) on posix: adding a separate testDependency module which implements main and provides app_start which could be used on posix would be fine I think.

bogdanm, 2015-12-18:
OK, that makes sense. `minar-platform-posix` already has a main, so it could also define `app-start`. It's not as clean as what you suggested (testDependency), but it'd probably work.

bogdanm, 2015-12-21:
Turns out that adding the definition of `app_start` into `minar-platform-posix` solves the problem only partially. If I try to break `core-util`'s dependency on `mbed-drivers` by moving the test environment into a separate repository, the dependency still needs to exist, since the definition of `app_start` currently lives in `mbed-drivers` (and of course `core-util` needs that definition for its tests). So either:

1. we move the definition of `app_start` into `core-util`, as originally proposed
2. we create another repository that contains the definition of `app_start` alone

I vote for 1, since 2 feels a bit too much.

bremoran, 2015-12-21:
That implies a dependency of core util on minar. That seems wrong, since minar is a consumer of core util. App start belongs in minar.

bogdanm, 2015-12-21:
That might be better indeed, since moving `app_start` alone is not enough, one also needs to move the `app_start` specific implementation of `main` out of `mbed-drivers`. `minar` does look like a good place for both of these, especially `minar` is the reason why we use `app_start` instead of `main` in mbed OS.

hugovincent, 2015-12-21:
@bogdanm true, but @autopulated made a good point above: putting `main` in `minar` itself complicates use of `minar` on POSIX or any non-mbed OS environment.

bogdanm, 2015-12-22:
@hugovincent, point taken, but this feels like we're worrying way too much about a bridge that we don't have to cross yet. In any case, do you (or anybody else) have a suggestion for a more elegant fix? What about this: we create a separate module (`mbed-app-start` ?) that contains both the definition of `app_start` and the mbed OS specific implementation of `main`. That module can be used both as a regular dependency (by `mbed-drivers` for example) or as a test dependency (when testing (say) `core-util` on POSIX). Would this work for you?

autopulated, 2015-12-22:
Sounds like core-util should have a testDependency on the implementation of main in mbed-drivers.

Note that you can easily reproduce the _declaration_ of app_start which is currently in `mbed-drivers/app.h`, if splitting that into a separate module seems unreasonable.

It also seems reasonable to me to have multiple implementations of main that do the same thing on different platforms: I don't see why you'd expect the mbed-os main-calls-app-start implementation to be the same as a posix equivalent (the mbed OS one possibly has to do some things the posix one doesn't, and visa versa).

bogdanm, 2015-12-22:
I don't think I understand your proposed solution here.

autopulated, 2015-12-22:
3. Write a new module that declares app_start and implements main() calling it, which is only used as a testTargetDependency when running on posix.

core-util should have a testTargetDependency on mbed-drivers

At some point in the future, our test framework grows to provide cross-platform abstraction of start/main, and modules have a testDependency on that, instead of on the implementation of main directly.

bogdanm, 2015-12-22:
> core-util should have a testTargetDependency on mbed-drivers

That's precisely what we're trying to avoid: a dependency on `mbed-drivers` for `core-util` or any other module that we want to be able to test in a POSIX environment. `mbed-drivers` doesn't plya nice on POSIX.

autopulated, 2015-12-22:
test**Target**Dependency

bogdanm, 2015-12-22:
OK, but you still need a place (a test target dependency) where you define `main` for Linux. Or would you rather define `main` separately in each module that needs to be testable in Linux?

autopulated, 2015-12-22:
> (3) Write a new module that declares app_start and implements main() calling it, which is only used as a testTargetDependency when running on posix.

bogdanm, 2015-12-22:
Sorry, I completely missed that initially. OK, I think we can do that.

autopulated, 2015-12-22:
no problem ;)

bogdanm, 2016-01-19:
Fixed by #137

closed 2016-01-19 by bogdanm