ARMmbed/mbed-drivers #125: Duplicate MAC address issue on K64F boards

devran01, 2015-11-25:
The function mbed_mac_address() defined in mbed-hal-frdm-k64f/source/mbed-overrides.c returns the mac address based on UUID. There is another definition with __weak attribute in mbed-drivers/source/mbed-interface.c that returns hardcoded mac address (0x00, 0x02, 0xF7, 0xF0, 0x00, 0x00). For one of the application developed in OpenAIS project, linker is always using definition with __weak attribute to resolve the symbol mbed_mac_address(), even though there is strong function defined in mbed-hal-frdm-k64f/source/mbed-overrides.c. Due to this default mac address being set on multiple K64F boards.

sg-, 2015-11-25:
This is due to .ar and and the linker. I recall an issue elsewhere but cant find it. Really, mbed_mac is legacy and the EthernetInterface should require the implementation of this. Quick and dirty solution would be to remove the weak implementation. Maybe this will get merged for the short term https://github.com/ARMmbed/mbed-drivers/pull/126

autopulated, 2015-11-25:
Yup removing the weak definition should fix this. I thought we had already done that, but apparently not!

0xc0170, 2015-11-25:
This issue is related to this problem: https://github.com/ARMmbed/mbed-hal-frdm-k64f/issues/3

sg-, 2015-11-25:
Thats the one! Thanks for tracking down :+1:

devran01, 2015-12-03:
Issue fixed in mbed-drivers 0.11.5.

closed 2015-12-03 by devran01

ansondtx20, 2016-01-06:
Using mbed-drivers 0.11.6 in https://github.com/ARMmbed/mbed-client-examples and I am having to manually set the MAC address in order to get DHCP (and network connectivity in general) to work... 

declared this above app_start(): 
extern "C" void mbed_mac_address(char *s) 
{
char mac[6];
mac[0] = 0x00;
mac[1] = 0x02;
mac[2] = 0xf7;
mac[3] = 0xf0;
mac[4] = 0x46;
mac[5] = 0x4b;
memcpy(s, mac, 6);
}

Then, in app_start(), put this in the very beginning:
void app_start(int /argc/, char* /argv/[]) {
// manually override the MAC address for now… 
char mmac[6];
mbed_mac_address(mmac);
…

Seemed to work.  Looks like there is something not quite right in 0.11.6