ARMmbed/mbed-drivers #132: Retarget - stdio baudrate (hal vs retarget) - elimination of duplication

0xc0170, 2015-12-11:
Each HAL implementation requires to implement code like this (taken from KSDK hal):

```   
#ifndef YOTTA_CFG_MBED_OS_STDIO_DEFAULT_BAUD
#   define YOTTA_CFG_MBED_OS_STDIO_DEFAULT_BAUD 9600
#endif

 if (obj->serial.instance == STDIO_UART) {
        UART_HAL_SetBaudRate(obj->serial.address, uartSourceClock, YOTTA_CFG_MBED_OS_STDIO_DEFAULT_BAUD);
        stdio_uart_inited = 1;
        memcpy(&stdio_uart, obj, sizeof(serial_t));
    }
```

There's init serial in retarget which is called only if stdio_uart_inited is not set to 1:

```
#ifndef YOTTA_CFG_MBED_OS_STDIO_DEFAULT_BAUD
#define YOTTA_CFG_MBED_OS_STDIO_DEFAULT_BAUD 115200
#endif

#define STDIO_DEFAULT_BAUD YOTTA_CFG_MBED_OS_STDIO_DEFAULT_BAUD

#if DEVICE_SERIAL
extern int stdio_uart_inited;
extern serial_t stdio_uart;
#endif

static void init_serial() {
#if DEVICE_SERIAL
    if (stdio_uart_inited) return;
    serial_init(&stdio_uart, STDIO_UART_TX, STDIO_UART_RX);
    serial_baud(&stdio_uart, STDIO_DEFAULT_BAUD);
#endif
}

```

Can we eliminate HAL duplication or minimize? I have just realized looking at the code above, YOTTA_CFG_MBED_OS_STDIO_DEFAULT_BAUD is not set to the same value if it's not defined (hal vs retarget),

closed 2016-01-19 by 0xc0170

lws-team, 2016-01-26:
Guys what are normal mortals meant to do using the public repo now?

```
ym/mbed-drivers/source/mbed-drivers.a(retarget.cpp.o): In function `init_serial':
/home/agreen/projects/mbed/lws-test-server/yotta_modules/mbed-drivers/source/retarget.cpp:98: undefined reference to `stdio_uart_inited'
/home/agreen/projects/mbed/lws-test-server/yotta_modules/mbed-drivers/source/retarget.cpp:98: undefined reference to `stdio_uart'
ym/mbed-drivers/source/mbed-drivers.a(retarget.cpp.o): In function `_write':
/home/agreen/projects/mbed/lws-test-server/yotta_modules/mbed-drivers/source/retarget.cpp:232: undefined reference to `stdio_uart_inited'
/home/agreen/projects/mbed/lws-test-server/yotta_modules/mbed-drivers/source/retarget.cpp:232: undefined reference to `stdio_uart'
ym/mbed-drivers/source/mbed-drivers.a(retarget.cpp.o): In function `_read':
/home/agreen/projects/mbed/lws-test-server/yotta_modules/mbed-drivers/source/retarget.cpp:259: undefined reference to `stdio_uart_inited'
/home/agreen/projects/mbed/lws-test-server/yotta_modules/mbed-drivers/source/retarget.cpp:259: undefined reference to `stdio_uart'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed
```

Shouldn't... you know... serial init always work for normal mortals?  I'm not talking about git HEAD.  It seems kinda basic...

0xc0170, 2016-01-26:
@lws-team 

Please report, the target + version, mbed-drivers version. Seems to me that you are using older mbed-drivers where these variables were declared as extern and defined in target's serial HAL. Those modules were updated.
I assume you fetched latest HAL for your target, but not mbed-drivers (as I recall both changes were patch versions). This could be avoided by more strict semver dependencie.

lws-team, 2016-01-26:
Hm I think you are right about mbed-drivers, but before posting I deleted yotta_modules/ and told mbed drivers dep was ^0.11.1, yotta update has fetched 0.11.1 from 'the public repository', I assume it is the latest in there.  And it is broken.  If nobody pushed new mbed-drivers needed to 'the public repository' stricter deps won't help.

If you're going to put this 'public repository' out there rather than fetch global sets of tested hashes from the truly public repository of github, it seems like it being broken is going to be a regular thing, this is the third time I updated and got a broken project.

I should solve this by replacing what yotta fetched with a checkput of the mbed-drivers git?

0xc0170, 2016-01-26:
The latest published mbed-drivers is 0.11.8 (http://yotta.mbed.com/#/module/mbed-drivers/0.11.8). Anyway, you are having in your dependency tree older mbed-drivers, but your target hal got updated to the latest patched version (where those uart definitions were removed, thus an error). What target are you using? I would like to know HAL tree modules.

> this is the third time I updated and got a broken project.

Sorry to hear that, this should not happen.

autopulated, 2016-01-26:
Since 0.13.0 `yotta outdated` will tell you why modules can't be updated (which module has a specification that's holding them back), hopefully this will help in these situations!

lws-team, 2016-01-26:
Yotta outdated told me I had ^0.11.1 in both the main module definition and the library definition apparently stopping it from using later versions (I got the wrong idea it seems ^ means same or later).  When I set these to 0.11.8, it can get 0.11.8 and build.  So thanks for the advice.
I removed all my ^ deps (there were only 3) and after yotta update
```
In file included from /home/agreen/projects/mbed/lws-test-server/yotta_modules/mbed-drivers/mbed-drivers/mbed.h:21:0,
                 from /home/agreen/projects/libwebsockets/lib/libwebsockets.h:29,
                 from /home/agreen/projects/libwebsockets/lib/private-libwebsockets.h:180,
                 from /home/agreen/projects/libwebsockets/lib/lws-plat-mbed3.cpp:1:
/home/agreen/projects/mbed/lws-test-server/yotta_modules/mbed-drivers/mbed-drivers/platform.h:21:20: fatal error: device.h: No such file or directory
compilation terminated.
```

device.h is there

./yotta_modules/mbed-hal-frdm-k64f/mbed-hal-frdm-k64f/device.h

yt outdated says

```
[agreen@build lws-test-server]$ yt outdated
websockets@1.6.0 not available from the registry
```
Yes that's true since this I don't know how to push to this "public respository".  But I have a git checkout of it in place by hand in yotta_modules.  Otherwise it thinks it has solved the deps.

yt target

```
[agreen@build lws-test-server]$ yt target
frdm-k64f-gcc 2.0.0
kinetis-k64-gcc 2.0.0
mbed-gcc 1.1.0
```

My "list" is like this atm

```
[agreen@build lws-test-server]$ yotta list
lws-test-server 1.5.0
┣━ websockets 1.6.0 yotta_modules/websockets -> /home/agreen/projects/libwebsockets
┣━ mbed-drivers 0.11.8
┃ ┣━ mbed-hal 1.2.1 yotta_modules/mbed-hal
┃ ┃ ┗━ mbed-hal-freescale 1.0.0 yotta_modules/mbed-hal-freescale
┃ ┃   ┗━ mbed-hal-ksdk-mcu 1.0.7 yotta_modules/mbed-hal-ksdk-mcu
┃ ┃     ┣━ uvisor-lib 1.0.12 yotta_modules/uvisor-lib
┃ ┃     ┗━ mbed-hal-k64f 1.0.1 yotta_modules/mbed-hal-k64f
┃ ┃       ┗━ mbed-hal-frdm-k64f 1.0.0 yotta_modules/mbed-hal-frdm-k64f
┃ ┣━ cmsis-core 1.1.1 yotta_modules/cmsis-core
┃ ┃ ┗━ cmsis-core-freescale 1.0.0 yotta_modules/cmsis-core-freescale
┃ ┃   ┗━ cmsis-core-k64f 1.0.0 yotta_modules/cmsis-core-k64f
┃ ┣━ ualloc 1.0.3 yotta_modules/ualloc
┃ ┃ ┗━ dlmalloc 1.0.0 yotta_modules/dlmalloc
┃ ┣━ minar 1.0.4 yotta_modules/minar
┃ ┃ ┗━ minar-platform 1.0.0 yotta_modules/minar-platform
┃ ┃   ┗━ minar-platform-mbed 1.0.0 yotta_modules/minar-platform-mbed
┃ ┣━ core-util 1.1.4 yotta_modules/core-util
┃ ┗━ compiler-polyfill 1.2.1 yotta_modules/compiler-polyfill
┗━ sockets 1.1.0
  ┗━ sal 1.1.2 yotta_modules/sal
    ┗━ sal-stack-lwip 1.1.3 yotta_modules/sal-stack-lwip
      ┣━ sal-driver-lwip-k64f-eth 1.0.3 yotta_modules/sal-driver-lwip-k64f-eth
      ┗━ sal-iface-eth 1.0.1 yotta_modules/sal-iface-eth
```

autopulated, 2016-01-26:
Ah I think the source of the problem here is that `^x.y.z` means "versions API-compatible with x.y.z", for `0.y.z` versions, this means the exact version only.

In general, I'd recommend using `^1.y.z` for modules with a stable API (1.x versions), and `~0.y.z` for 0.x versions. `~x.y.z` always means that the patch version is allowed to vary, which for 0.x versions means that things may potentially break, but allows enough flexibility to update to at least some new versions.

lws-team, 2016-01-26:
That may have been the source of the original problem but my local deps look like this (and looked like it at the time of my previous post)

```
[agreen@build lws-test-server]$ cat module.json 
{
  "name": "lws-test-server",
  "version": "1.5.0",
  "bin": "./source",
  "description": "Libwebsockets test server",
  "author": "Andy Green <andy@warmcat.com>",
  "license": "Apache-2.0",
  "extraIncludes": [ "build/frdm-k64f-gcc/ym/websockets/existing" ],
  "dependencies": {
	"websockets": "^1.6.0",
	"mbed-drivers": ""
  }
}
```

and for the library

```
[agreen@build lws-test-server]$ cat yotta_modules/websockets/module.json 
{
  "name": "websockets",
  "version": "1.6.0",
  "description": "Libwebsockets",
  "keywords": [
    "lws",
    "libwebsockets",
    "websockets",
    "ws"
  ],
  "author": "Andy Green <andy@warmcat.com>",
  "homepage": "https://libwebsockets.org",
  "license": "LGPL2.1-SLE",
  
  "extraIncludes": [ "build/frdm-k64f-gcc/generated/include" ],
  "dependencies": {
  }
}
```

otherwise everything is managed by yotta / "the public repository".

I shouldn't have to do anything here to make it find its own device.h, right?

autopulated, 2016-01-26:
No you shouldn't. Does that error persist after `yotta clean` / removing the build directory?

lws-team, 2016-01-26:
I could get it to build again with:

```
[agreen@build lws-test-server]$ cat module.json 
{
  "name": "lws-test-server",
  "version": "1.5.0",
  "bin": "./source",
  "description": "Libwebsockets test server",
  "author": "Andy Green <andy@warmcat.com>",
  "license": "Apache-2.0",
  "extraIncludes": [ "build/frdm-k64f-gcc/ym/websockets/existing" ],
  "dependencies": {
	"websockets": "^1.6.0",
	"mbed-drivers": "",
	"sockets": ""
  }
}
```
and
```
[agreen@build lws-test-server]$ cat yotta_modules/websockets/module.json 
{
  "name": "websockets",
  "version": "1.6.0",
  "description": "Libwebsockets",
  "keywords": [
    "lws",
    "libwebsockets",
    "websockets",
    "ws"
  ],
  "author": "Andy Green <andy@warmcat.com>",
  "homepage": "https://libwebsockets.org",
  "license": "LGPL2.1-SLE",
  
  "extraIncludes": [ "build/frdm-k64f-gcc/generated/include" ],
  "dependencies": {
    "mbed-drivers": "",
    "sal-stack-lwip": "",
    "sockets": ""
  }
}
```

But the thing I was testing, DHCP on K64F from the "public repository" is still broken, it just times out.  This worked a few weeks ago...  anyway I'll go moan on that repo.

Thanks for the help.