ARMmbed/mbed-drivers #134: mbed-drivers - error missing cstddef

0xc0170, 2015-12-15:
This is the line which fails to build for some users when using ubuntu, or even on macs. 

Found reported in various places, various "fixes-" but all seemed to be effective, here are some references: 
http://forums.mbed.com/t/yotta-on-raspberry-pi-2/842/2 (reported with yotta + mbed OS)
https://github.com/tmk/tmk_keyboard/issues/212 (mbed 2 sdk, but same error with the same header, provides some additional information)
https://developer.mbed.org/forum/mbed/topic/5205 (I saw this first time back in 2014 in this thread)

@jaustin Any pointers?

jaustin, 2015-12-15:
I think this is a malformed toolchain... I wonder if all these users are on Debian?

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=736167

Seems to fit, as well as with comments like "I also had similar problem with GCC4.8 on Ubuntu 14.04.
I had to update to 4.9 with this PPA." and " everything is okay now, had to take the tools for trusty." from the other reports.

I guess we could test for the include and give some more helpful advice?

J

autopulated, 2015-12-15:
@jaustin we're wondering if there's an issues that still needs to be fixed upstream somewhere beyond the "everything is okay now, had to take the tools for trusty.", as people are definitely still experiencing this. Maybe they are on old versions though.

bogdanm, 2016-01-08:
This looks indeed like a toolchain issue. I'd like to try replacing `#include <cstddef>` with '#include <stddef.h>` and see if that does any good, but I don't know how to reproduce the problem on my machine. Suggestions welcome.

rgrover, 2016-01-28:
I also encountered this build failure on my Ubuntu (Ubuntu 14.04.3 LTS) using arm-gcc v4.9.3. I'll try to upgrade the toolchain.

rgrover, 2016-02-01:
Updating the toolchain following instructions from [GCC ARM Embedded](https://launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa); this fixed the issue for me.