ARMmbed/mbed-drivers #135: Test heap and stack - warning - does not iterate

0xc0170, 2016-01-04:
The test is outdated, as it assumes the stack is above the heap (and smaller than heap).

frdm-k64f-gcc output:
```
{{timeout;10}}
{{host_test_name;default_auto}}
{{description;Heap & Stack}}
{{test_id;MBED_13}}
{{start}}
Initial stack/heap geometry:
   stack pointer:V 0x1fffa047
   heap pointer :^ 0x1fffacb0

stack pointer: 0x1fffa02f
heap pointer: 0x1fffadc0

[WARNING] The Stack/Heap collision was not detected

Allocated (0)Kb in (1) iterations

{{success}}
{{end}}
```

This test should be in ualloc, and should be corrected/improved.

Thanks for bringing this up to @akselsm

PrzemekWirkus, 2016-02-23:
This test will be removed from test async test collection for mbed-drivers.
@0xc0170 I think we can close it ?

0xc0170, 2016-02-23:
Yes, shall be removed. It will be closed once the test is not in the new version.