ARMmbed/mbed-drivers #136: Why do we need app.h?

bogdanm, 2016-01-07:
I'm trying to figure out why we needed https://github.com/ARMmbed/mbed-drivers/blob/master/mbed-drivers/app.h in the first place. It declares `app_start` with C  linkage, which seem pointless, since MINAR is not designed to be used from C code. If we eliminate the C linkage, we don't strictly need this file anymore, which is an advantage is we run mbed OS in another environment (for example Linux) and we want to avoid including `mbed-drivers` and implicitly `app.h` (which is the main reason for opening this issue). I vote to replace the contents of this file with a wawrning stating that it is obsolete and it should not be used, and nothing more.
Thoughts? @bremoran @0xc0170

0xc0170, 2016-01-07:
Wasn't the solution proposed in #118? As I read the last comments there, I would say yes?

> I vote to replace the contents of this file with a wawrning stating that it is obsolete and it should not be used, and nothing more.

Where the app would be declared? The function needs a declaration somewhere to declare it in the scope (in C++).

bogdanm, 2016-01-07:
> Wasn't the solution proposed in #118?

That's more a solution for implementing the mbedOS `main` based function in a non-embedded environment, not as much for `app.h`

> Where the app would be declared? The function needs a declaration somewhere to declare it in the scope (in C++). 

Strictly speaking, it doesn't. You can just declare it and the linker will happily use it if it has the right signature or issue an error if it doesn't.

0xc0170, 2016-01-07:
> Strictly speaking, it doesn't. You can just declare it and the linker will happily use it if it has the right signature or issue an error if it doesn't.

That's what I was asking, where would it be declared. The proposal as I understand it:

```
//retarget.cpp

void app_start(int argc, char *argv[]);

extern "C" int main(void) {
    minar::Scheduler::postCallback(
        mbed::util::FunctionPointer2<void, int, char**>(&app_start).bind(0, NULL)
    );
    return minar::Scheduler::start();
}

```

```
// header app.h

#ifndef __MBED_CORE_MBED_APP_H__
#define __MBED_CORE_MBED_APP_H__

#warning Deprecated......

#endif // ndef __MBED_CORE_MBED_APP_H__
```

bogdanm, 2016-01-07:
Yes, that is the proposal. We keep `app.h` around for a while to ensure compatibility with previous code, but it won't actually do anything anymore. Then we declare `app_start` in retarget.cpp, just as in the fragment you posted above, and we implement it directly wherever it's needed (application code or test).

autopulated, 2016-01-07:
It was declared with C linkage to be most similar to `main`, which always has C linkage, but I think this is a good idea.

0xc0170, 2016-01-08:
> I think this is a good idea

The proposal? +1 from me

closed 2016-01-29 by 0xc0170