ARMmbed/mbed-drivers #140: Forcing serial to be enabled in retarget.cpp prevents the NRF51 from entering low power mode.

marcuschangarm, 2016-01-15:
The sleep current jumps from ~1uA to ~4 mA.

https://github.com/ARMmbed/mbed-drivers/pull/139

0xc0170, 2016-01-18:
Please provide more details of the testing app, your target , toolchain.

I assume your app does not use any Serial object, neither stdio. Previously with gcc toolchain, the serial stdio object would not be initialized until open for stdin/out/error was invoked. This was not case for ARMCC. Did you test ARMCC ? The fix you referenced was a fix for regression casued by the previous pull request for stdio serial, thus we will need to look at both change-sets.

marcuschangarm, 2016-01-18:
This is a feature on the NRF51. If you touch the serial, you use more power. 

My question to you is why on Earth would you initialize the serial port by default for all mbed targets? Once you start building actual products the serial line is most likely not used at all. On some targets, setting the USBTX/RX to NC and calling init_serial results in a hardfault, for who would initialize a peripheral that is not connected?

bogdanm, 2016-01-18:
If you look at the code:

https://github.com/ARMmbed/mbed-drivers/blob/master/source/retarget.cpp#L109

you'll see that the actual initialization only happens for those targets that have `DEVICE_SERIAL` enabled.

marcuschangarm, 2016-01-18:
Setting ```DEVICE_SERIAL``` to 0 breaks on the NRF51 and EFM32. It works on the K64F.

I thought the ```DEVICE_SERIAL``` meant whether or not the peripheral was present on the platform, not if you want to use it in you application or not?

PR #145 (2016-01-29): Retarget - stdio serial lazy initialization
We provide mbed::get_serial_stdio() to get a reference for stdio Serial
object. We should discourage to use Serial pc(USBTX, USBRX)

We should clean our tests/examples, to use this new function for stdio Serial operations like baud(), format() or any other.

Should fix #140

@marcuschangarm @bogdanm @pan- @bremoran
closed 2016-01-29 by 0xc0170