ARMmbed/mbed-drivers #142: Default Baud rate

bridadan, 2016-01-19:
Greentea defaults to a baud of 115200 now but mbed-drivers defaults to 9600: https://github.com/ARMmbed/mbed-drivers/blob/master/source/SerialBase.cpp#L29

These need to be reconciled because it causes new yotta targets to fail when running `yotta test` during the porting process.

Should mbed-drivers be bumped up to 115200? This would require lots of documentation updates and warnings since it has been 9600 for so long. Or should we lower Greentea?

Not sure how to proceed, so hoping to start a discussing here.

bogdanm, 2016-01-19:
We made it 9600 initially precisely because of greentea issues: https://github.com/ARMmbed/target-mbed-gcc/commit/bd2198b909d3a24290713b0b9f26231efa36eaae

The real solution to this, as described in the link above, is for greentea to read `yotta config` data. Not sure what's the status of that.

autopulated, 2016-01-20:
greentea reading yotta config sounds a bit wrong: isn't it possible to standardise on a baud rate for tests?

0xc0170, 2016-01-20:
> The real solution to this, as described in the link above, is for greentea to read yotta config data. Not sure what's the status of that.

Greentea already reads the config to get baudrate, it was integrated a while ago ( I can look up the PR there if needed). The SerialBase defaults to 9600, as it can be any serial module (== or != stdin/out), plus many of our HAL currently do the same, hardcode to 9600 in serial_init().

> it causes new yotta targets to fail when running yotta test during the porting process.

mbed targets (mbed-gcc and mbed-armcc) set default-baudrate to 9600 for compatibility, https://github.com/ARMmbed/target-mbed-gcc/blob/master/target.json#L24. Should not fail, more details please. What problems are you having? The test does not use Serial object, or are there tests which overwrite stdio by creating Serial on that port to change baudrate ? I am aware of one issue created in either ualloc or somewhere (have to look at it to fix it).

> greentea reading yotta config sounds a bit wrong: isn't it possible to standardise on a baud rate for tests?

This might be further discussed in greentea as it's already implemented. Can we standardize, yes to 9600, no? There are tests which can print a lot of information (I used to have a socket test which would take 60s with 9600, 15s without those debug messages).

hugovincent, 2016-01-25:
> isn't it possible to standardise on a baud rate for tests?

There is also the problem that some targets don't work at 115kbaud under all relevant clock configurations and workloads (e.g. Nordic nRF51 when the BT stack is running needs flow control at 115kbaud). Conversely, some of the mesh tests can't run at 9600 baud. My vote: standardise on 115kbaud anyway, and work around in the handful of cases where it's a problem.

mlnx, 2016-01-25:
+1
for standardizing to 115kbaud

bridadan, 2016-01-26:
Closing this for now as it seems to be an issue specifically with `yotta test` and not the drivers themselves (the tests work fine with `mbedgt`

closed 2016-01-26 by bridadan