ARMmbed/mbed-drivers #143: Do Tests Work?

jacobrosenthal, 2016-01-21:
Im following the greentea guide to test mbed drivers on my (custom) target and they all fail:
Im clearly an outsider to greentea and mbed so it could easily be some misunderstanding I have.
The error seems to be 
```
HOST: Error! No host test name defined in preamble
```
Which I think is related to the `MBED_HOSTTEST_SELECT(default_auto);` declaration in all tests?

Is it my target, and do tests actually pass right now? or did something break upstream in testing? 

```
Jacobs-MacBook-Air:mbed-drivers jacobrosenthal$ mbedgt -V
mbedgt: checking for yotta target in current directory
	reason: no --target switch set
mbedgt: checking yotta target in current directory
	calling yotta: yotta --plain target
mbedgt: assuming default target as 'tinyble-gcc'
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
	detected 'ARCH_BLE' -> 'ARCH_BLE[0]', console at '/dev/tty.usbmodem1422', mounted at '/Volumes/MBED', target id '9009020305B9782BF84517DA'
mbedgt: local yotta target search in './yotta_targets' for compatible mbed-target 'arch_ble'
	inside './yotta_targets/tinyble-gcc' found compatible target 'tinyble-gcc'
mbedgt: processing 'tinyble-gcc' yotta target compatible platforms...
mbedgt: processing 'ARCH_BLE' platform...
mbedgt: using platform 'ARCH_BLE' for test:
	target_id_mbed_htm = '9009020305B9782BF84517DA'
	mount_point = '/Volumes/MBED'
	target_id = '9009020305B9782BF84517DA'
	serial_port = '/dev/tty.usbmodem1422'
	target_id_usb_id = '9009020305B9782BF84517DA'
	platform_name = 'ARCH_BLE'
	platform_name_unique = 'ARCH_BLE[0]'
mbedgt: building your sources and tests with yotta...
	calling yotta: yotta --target=tinyble-gcc,* build
info: generate for target: tinyble-gcc 0.0.5 at /Users/jacobrosenthal/Downloads/mbed-drivers/yotta_targets/tinyble-gcc
GCC version is: 4.8.4
suppressing warnings from nrf51-sdk
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/jacobrosenthal/Downloads/mbed-drivers/build/tinyble-gcc
ninja: no work to do.
mbedgt: yotta build for target 'tinyble-gcc' was successful
mbedgt: running 20 tests for target 'tinyble-gcc' and platform 'ARCH_BLE'
	use 1 instance for testing
mbedgt: selecting test case observer...
	calling mbedhtrun: mbedhtrun -d /Volumes/MBED -p /dev/tty.usbmodem1422:9600 -f "./build/tinyble-gcc/test/mbed-drivers-test-serial_interrupt.bin" -C 6 -c shell -m ARCH_BLE
mbedgt: mbed-host-test-runner: started
HOST: Instrumentation: "/dev/tty.usbmodem1422" and disk: "/Volumes/MBED"
HOST: My PID is 5236
HOST: Copy image onto target...
HOST: Initialize serial port...
...port ready!
HOST: Reset target...
HOST: Detecting test case properties...
HOST: Error! No host test name defined in preamble
```

0xc0170, 2016-01-21:
Yes they should work.  I retested now with mbedgt 0.1.15, frdm-k64f-gcc target.

```
# snippet of my cmd
[168/168] Linking CXX executable test/mbed-drivers-test-timeout                                                                             
mbedgt: yotta build for target 'frdm-k64f-gcc' was successful                                                                               
mbedgt: running 20 tests for target 'frdm-k64f-gcc' and platform 'K64F'                                                                     
        use 1 instance for testing                                                                                                          
mbedgt: selecting test case observer...                                                                                                     
        calling mbedhtrun: mbedhtrun -d F: -p COM64:9600 -f ".\build\frdm-k64f-gcc\test\mbed-drivers-test-serial_interrupt.bin" -C 4 -c shel
l -m K64F                                                                                                                                   
mbedgt: mbed-host-test-runner: started                                                                                                      
MBED: Instrumentation: "COM64" and disk: "F:"                                                                                               
HOST: Copy image onto target...                                                                                                             
        1 file(s) copied.                                                                                                                   
HOST: Initialize serial port...                                                                                                             
...port ready!                                                                                                                              
HOST: Reset target...                                                                                                                       
HOST: Detecting test case properties...                                                                                                     
HOST: Property 'timeout' = '20'                                                                                                             
HOST: Property 'host_test_name' = 'echo'                                                                                                    
HOST: Property 'description' = 'serial interrupt test'                                                                                      
HOST: Property 'test_id' = 'MBED_14'                                                                                                        
```

Please share the version of your tools (mbedgt, htrun). You are using ARCH_BLE, does that target inherit from mbed target ?

bridadan, 2016-01-21:
I believe this is related to #142. Try adding the following to your `target.json`'s config section:

```
"config": {
  "mbed-os": {
    "stdio": {
      "default-baud": 115200
    }
  }
}
```

jacobrosenthal, 2016-01-21:
Added that to config, and it appears to be using 115200 now, but still fails with the same error:

```
Jacobs-MacBook-Air:mbed-drivers jacobrosenthal$ mbedgt -V
mbedgt: checking for yotta target in current directory
	reason: no --target switch set
mbedgt: checking yotta target in current directory
	calling yotta: yotta --plain target
mbedgt: assuming default target as 'tinyble-gcc'
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
	detected 'ARCH_BLE' -> 'ARCH_BLE[0]', console at '/dev/tty.usbmodem1422', mounted at '/Volumes/MBED', target id '9009020305B9782BF84517DA'
mbedgt: local yotta target search in './yotta_targets' for compatible mbed-target 'arch_ble'
	inside './yotta_targets/tinyble-gcc' found compatible target 'tinyble-gcc'
mbedgt: processing 'tinyble-gcc' yotta target compatible platforms...
mbedgt: processing 'ARCH_BLE' platform...
mbedgt: using platform 'ARCH_BLE' for test:
	target_id_mbed_htm = '9009020305B9782BF84517DA'
	mount_point = '/Volumes/MBED'
	target_id = '9009020305B9782BF84517DA'
	serial_port = '/dev/tty.usbmodem1422'
	target_id_usb_id = '9009020305B9782BF84517DA'
	platform_name = 'ARCH_BLE'
	platform_name_unique = 'ARCH_BLE[0]'
mbedgt: building your sources and tests with yotta...
	calling yotta: yotta --target=tinyble-gcc,* build
info: generate for target: tinyble-gcc 0.0.5 at /Users/jacobrosenthal/Downloads/mbed-drivers/yotta_targets/tinyble-gcc
GCC version is: 4.8.4
suppressing warnings from nrf51-sdk
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/jacobrosenthal/Downloads/mbed-drivers/build/tinyble-gcc
ninja: no work to do.
mbedgt: yotta build for target 'tinyble-gcc' was successful
mbedgt: running 20 tests for target 'tinyble-gcc' and platform 'ARCH_BLE'
	use 1 instance for testing
mbedgt: selecting test case observer...
	calling mbedhtrun: mbedhtrun -d /Volumes/MBED -p /dev/tty.usbmodem1422:115200 -f "./build/tinyble-gcc/test/mbed-drivers-test-serial_interrupt.bin" -C 6 -c shell -m ARCH_BLE
mbedgt: mbed-host-test-runner: started
HOST: Instrumentation: "/dev/tty.usbmodem1422" and disk: "/Volumes/MBED"
HOST: My PID is 49032
HOST: Copy image onto target...
HOST: Initialize serial port...
...port ready!
HOST: Reset target...
HOST: Detecting test case properties...
HOST: Error! No host test name defined in preamble

{{error}}
{{end}}
Hmbedgt: mbed-host-test-runner: stopped

```

0xc0170, 2016-01-22:
@jacobrosenthal what system are you using? This seems to be a problem with greentea/htrun. Please add versions of mbedgt, mbedhtrun, mbedls and your setup (version of your operating system). What does mbedls report?

Please try this example to run, https://github.com/ARMmbed/greentea#example-1---digest-the-default-mbed-host-test-runner

0xc0170, 2016-01-22:
For instance, this is one of the issues which might be related: https://github.com/ARMmbed/htrun/issues/2 . As you are using mbedgt, report a problem there please.

0xc0170, 2016-01-26:
@jacobrosenthal Any update? did you report this to mbedgt?

closed 2016-01-27 by jacobrosenthal