ARMmbed/mbed-drivers #144: Declaring PwmOut makes the board not startup anymore

janjongboom, 2016-01-21:
I have a FRDM-K64F board, tri-color LED hooked up on D5, D6, D7. Another LED hooked up on D8. Now when I declare D8 as PwmOut my application does not start up anymore, app_start() is not hit, and no output over Serial either...

```cpp
static DigitalOut red(YOTTA_CFG_HARDWARE_PINS_D5);
static DigitalOut blue(YOTTA_CFG_HARDWARE_PINS_D6);
static DigitalOut green(YOTTA_CFG_HARDWARE_PINS_D7);

static PwmOut green2(YOTTA_CFG_HARDWARE_PINS_D8);

void app_start(int /*argc*/, char* /*argv*/[]) {
    blue = true;
    red = true;
}
```

If you comment the PwmOut line out, everything works again...

0xc0170, 2016-01-21:
@janjongboom What is your revision for K64F. PTA8 for D8 was for earlier revisions which was wrong as this is OpenSDA pin (SWD_CLK_TGTMCU).

This is the issue for FRDM-K64F target, which should define revision and fix this D8 define for different revisions.

More information: https://github.com/mbedmicro/mbed/issues/1035

0xc0170, 2016-01-21:
Please create an issue for frdm-k64f target: https://github.com/ARMmbed/target-frdm-k64f-gcc (armcc will need this fix as well).

this is the problematic line: https://github.com/ARMmbed/target-frdm-k64f-gcc/blob/master/target.json#L51

closed 2016-01-21 by 0xc0170