ARMmbed/mbed-drivers #150: Default baudrate default serial STDIO object changed to 9600

kmdewaal, 2016-02-03:
Default baudrate used to be 115200, as documented. Since mbed-drivers version 0.12.0 the default baudrate of the serial STDIO object, the one that is used by the printf statements, has changed to 9600.

0xc0170, 2016-02-03:
@kmdewaal What target are you using? The top parent targets have it set to 9600 in their configs, an example : https://github.com/ARMmbed/target-mbed-gcc/blob/master/target.json#L24

This default if its not set was not changed, https://github.com/ARMmbed/mbed-drivers/blob/master/source/retarget.cpp#L62.

Please provide more details, even a code snippet which we can test  plus a target, specific mbed-drivers version which worked for you with 115k

kmdewaal, 2016-02-03:
OK, I do understand now that the default baud rate is intended to be 9600 and not 115200 and that the correct way to set the baud rate is with the get_stdio_serial() function as documented in https://github.com/ARMmbed/mbed-drivers.

My application contained the following code to set the baud rate to 115200:

> static Serial pc(USBTX, USBRX);
> pc.baud(115200);  //Setting the Baud-Rate for trace output
> 

This code did set the baud rate to 115200 until last Friday. After a "yotta update" in which the mbed-drivers version went up to 0.12.0 the actual baud rate was 9600; the code to set the baud rate to 115200 had no effect anymore on the "printf()" output. However, the output of "pc.printf()" was at a baud rate of 115200.
This lead me to the conclusions that (a) my original code did not work at all on the default STDIO and that (b) the default baud rate of the STDIO was changed from 115200 to 9600 as of last Friday; hence the issue. I do apologize for the wrong conclusions!
I do understand now that (a) my original code did work but now not anymore and (b) the default baud rate was and is 9600.

After reading the documentation as in https://github.com/ARMmbed/mbed-drivers I have modified my application accordingly:

> Serial &pc = get_stdio_serial();
> pc.baud(115200);  //Setting the Baud-Rate for trace output

This does actually set the baud rate to 115200.

Thanks for the quick response and I consider the issue closed.

closed 2016-02-03 by kmdewaal

0xc0170, 2016-02-03:
Thanks for sharing more details. yes, the latest version added a new function to get stdio get object, to address the issues with creating a new object (Serial pc(tx,rx) for changing the stdio properties and mismatch between yotta config and runtime.