ARMmbed/mbed-drivers #154: Printing stack trace on crash.

javier-moreno-tridonic-com, 2016-02-19:
I don't know if this is the proper repo to open this issue. But since I assume that mbed_die is intended to crash the program on demand here it goes.

From our personal experience we found out that for debugging purposes and long term test, will be very interesting to be able to log the stack trace in case of error or crash.

Typically on unix systems that can be achieved with "execinfo.h" library. But AFAIK there is no such library available in the GNU tools for ARM embedded.

Does mbedos provides a mean to achieve this? If not, is there anything planned or a way to achieve it?

0xc0170, 2016-02-19:
cc @AlessandroA

meriac, 2016-02-22:
We have an open issue where we collect stakeholders for generalised fault handling at https://github.com/ARMmbed/uvisor/issues/70: In case it covers your requirements, feel free to close this ticket as a duplicate.

javier-moreno-tridonic-com, 2016-02-23:
Well initially we are not using uvisor (we even get a warning that is not supported for our current target). So initially we were looking for something simpler. Will uvisor be a requirement to use the debug suite?

meriac, 2016-02-23:
@javier-moreno-tridonic-com No, what I propose is a uvisor independent driver that maps debug outputs / stack traces to whatever hardware you have. For many devices this simply means UART debug output. uVisor would just utilise that driver for solving the same problem you have - but you should be fine with using the same driver with some OS-specific code in case you don't have uVisor around.

javier-moreno-tridonic-com, 2016-02-24:
I see. That will fit our requirements. Thanks for the explanation.
Closing this as a duplicate then.

closed 2016-02-24 by javier-moreno-tridonic-com