ARMmbed/mbed-drivers #166: V2::I2C not compatible with Silicon Labs boards

marcuschangarm, 2016-03-02:
From https://github.com/ARMmbed/mbed-drivers/blob/master/source/v2/I2C.cpp#L122-L131

```
I2C::I2C(PinName sda, PinName scl) :
    _hz(100000)
{
    // Select the appropriate I2C Resource Manager
    uint32_t i2c_sda = pinmap_peripheral(sda, PinMap_I2C_SDA);
    uint32_t i2c_scl = pinmap_peripheral(scl, PinMap_I2C_SCL);
    int ownerID = pinmap_merge(i2c_sda, i2c_scl);
    _owner = detail::get_i2c_owner(ownerID);
    CORE_UTIL_ASSERT(I2CError::None == _owner->init(sda, scl));
}
```

On the Silicon Labs boards, ```i2c_sda``` and ```i2c_scl``` are defined as the memory address where they can be accessed. This breaks the I2C driver, because ```get_i2c_owner``` assumes the merged addresses, ```ownerID```, can be used as an array index directly: https://github.com/ARMmbed/mbed-drivers/blob/master/source/v2/I2CDetail.cpp#L306

Also, when debug is disabled, the following line is removed and ```init``` is not called:

```CORE_UTIL_ASSERT(I2CError::None == _owner->init(sda, scl));```

@bremoran

0xc0170, 2016-03-03:
@marcuschangarm Thanks for reporting. I'll send fix for that assert.

We overlooked that pinmap_merge result, that it might not be always a module number. To add a bit more details, taken from efm32 HAL:

```
obj->i2c = (I2C_TypeDef*) pinmap_merge(i2c_sda, i2c_scl);
```

nrf51 does not use pinmap_merge, here's a code snippet from its i2c init: 

```
    // Initialize variable to avoid compiler warnings
    NRF_TWI_Type *i2c = (NRF_TWI_Type *)I2C_0;

    if (i2c0_spi0_peripheral.usage == I2C_SPI_PERIPHERAL_FOR_I2C &&
            i2c0_spi0_peripheral.sda_mosi == (uint8_t)sda &&
            i2c0_spi0_peripheral.scl_miso == (uint8_t)scl) {
        // The I2C with the same pins is already initialized
        i2c = (NRF_TWI_Type *)I2C_0;
        obj->peripheral = 0x1;
    } else if (i2c1_spi1_peripheral.usage == I2C_SPI_PERIPHERAL_FOR_I2C &&
            i2c1_spi1_peripheral.sda_mosi == (uint8_t)sda &&
            i2c1_spi1_peripheral.scl_miso == (uint8_t)scl) {
        // The I2C with the same pins is already initialized
        i2c = (NRF_TWI_Type *)I2C_1;
        obj->peripheral = 0x2;
    } else if (i2c0_spi0_peripheral.usage == 0) {
        i2c0_spi0_peripheral.usage = I2C_SPI_PERIPHERAL_FOR_I2C;
        i2c0_spi0_peripheral.sda_mosi = (uint8_t)sda;
        i2c0_spi0_peripheral.scl_miso = (uint8_t)scl;

        i2c = (NRF_TWI_Type *)I2C_0;
        obj->peripheral = 0x1;
    } else if (i2c1_spi1_peripheral.usage == 0) {
        i2c1_spi1_peripheral.usage = I2C_SPI_PERIPHERAL_FOR_I2C;
        i2c1_spi1_peripheral.sda_mosi = (uint8_t)sda;
        i2c1_spi1_peripheral.scl_miso = (uint8_t)scl;

        i2c = (NRF_TWI_Type *)I2C_1;
        obj->peripheral = 0x2;
    } else {
        // No available peripheral
        error("No available I2C");
    }
```

akselsm, 2016-03-03:
Yeah, to expand on this, our [I2CName typedef](https://github.com/ARMmbed/mbed-hal-silabs/blob/master/mbed-hal-efm32/PeripheralNames.h#L47-L56) is defined as

```C
typedef enum {
#if defined(I2C0_BASE)
    I2C_0 = I2C0_BASE,
#endif
#if defined(I2C1_BASE)
    I2C_1 = I2C1_BASE,
#endif
} I2CName;
```

You cannot assume that enums start counting from zero. In general, since there is absolutely no documentation on how the HAL is expected to behave, apart from supplying implementations for the prototypes in mbed-hal, I would be **very** hesitant to start using things that historically have been internal to the HAL without first going through and actually specifying the interface, and then giving vendors time to implement what is, in practice, a new version of the HAL.

bremoran, 2016-03-03:
It looks like we require a new API in the I2C HAL that will be equivalent to the one in SPI:

```C
/** Get the module number
 *
 * @param[in] obj The SPI peripheral to check
 * @return The module number
 */
uint8_t spi_get_module(spi_t *obj);
```

marcuschangarm, 2016-03-03:
```
    int ownerID = pinmap_merge(i2c_sda, i2c_scl);
    _owner = detail::get_i2c_owner(0);
```

These functions are only called once per instantiation. The easiest thing to do is count the peripherals in PinMap_I2C_SDA on each call and return the index based on that.

PR #167 (2016-03-03): I2C - init() call out of the assert
Partly fixes #166 . However, as expected, if CORE_UTIL_ASSERT is not active (NDEBUG defined), this results in the warning:

`` warning: unused variable 'result' [-Wunused-variable] I2CError result = _owner->init(sda, scl); ``

We might want to use error() for this:

```
    if (I2CError::None == _owner->init(sda, scl)) {
        error("I2C init failed with an error");
    }
```

@bogdanm @bremoran
PR #173 (2016-03-07): Add a target-independent instance selector
Selects the instance by finding unique peripheral IDs in the pinmap.

Fixes #166 

cc @0xc0170 @marcuschangarm @bogdanm
commit e3af6a44 (2016-03-07): Add a target-independent instance selector
This should fix #166
closed 2016-03-07 by bremoran