ARMmbed/mbed-drivers #168: source/v2/I2C.cpp:22:28: fatal error: PeripheralPins.h: No such file or directory

bearsh, 2016-03-03:
PeripheralPins.h is not existent in configuration. I suppose it should be hard of the target cpu hal...
```
my-ble 0.0.0
┣━ ble 2.5.2
┃ ┣━ ble-nrf51822 2.5.3 yotta_modules/ble-nrf51822
┃ ┃ ┗━ nrf51-sdk 2.2.1 yotta_modules/nrf51-sdk
┃ ┣━ mbed-drivers 1.1.2 yotta_modules/mbed-drivers
┃ ┃ ┣━ mbed-hal 1.2.2 yotta_modules/mbed-hal
┃ ┃ ┃ ┗━ mbed-hal-nordic 2.0.0 yotta_modules/mbed-hal-nordic
┃ ┃ ┃   ┗━ mbed-hal-nrf51822-mcu 2.1.6 yotta_modules/mbed-hal-nrf51822-mcu
┃ ┃ ┃     ┗━ mbed-hal-nrf51dk 2.0.0 yotta_modules/mbed-hal-nrf51dk
┃ ┃ ┣━ cmsis-core 1.1.2 yotta_modules/cmsis-core
┃ ┃ ┃ ┗━ cmsis-core-nordic 1.0.1 yotta_modules/cmsis-core-nordic
┃ ┃ ┃   ┗━ cmsis-core-nrf51822 1.3.2 yotta_modules/cmsis-core-nrf51822
┃ ┃ ┣━ ualloc 1.1.0 yotta_modules/ualloc
┃ ┃ ┃ ┗━ dlmalloc 1.0.0 yotta_modules/dlmalloc
┃ ┃ ┣━ minar 1.0.4 yotta_modules/minar
┃ ┃ ┃ ┗━ minar-platform 1.0.0 yotta_modules/minar-platform
┃ ┃ ┃   ┗━ minar-platform-mbed 1.1.2 yotta_modules/minar-platform-mbed
┃ ┃ ┗━ core-util 1.5.3 yotta_modules/core-util
┃ ┗━ compiler-polyfill 1.2.1 yotta_modules/compiler-polyfill
┗━ z85 0.2.2
```

0xc0170, 2016-03-03:
@bearsh Thanks for reporting. We are already working on the fix.

adbridge, 2016-03-07:
#169 Fixes this issue

closed 2016-03-07 by 0xc0170