ARMmbed/mbed-drivers #172: Release process when changing dependency versions

bridadan, 2016-03-03:
There was a recent issue with the core-util module having a circular dependency on an older version of mbed-drivers. This lead to broken examples during a hackathon and workshop that was given over the weekend.

These dependency problems can be easily avoided in the future by having basic CI checks. Are these tests currently in place? If so, how come the change it made it through?

0xc0170, 2016-03-04:
@PrzemekWirkus

bridadan, 2016-03-09:
Anybody?

As an experiment, I made a Travis CI config that will build all the tests whenever a PR/commit is made. Do we need something like this? Or is this being handled internally?

Link to the experiment: https://github.com/bridadan/mbed-drivers/pull/1

0xc0170, 2016-03-10:
@bridadan +1 to have the travis running automatically.

bridadan, 2016-03-10:
Closing this, let's continue any discussion on the PR #180

closed 2016-03-10 by bridadan