ARMmbed/mbed-drivers #174: add config option to disable retargeting

bearsh, 2016-03-04:
retarget always init the serial port as soon as `DEVICE_SERIAL` is set. to me that's a strange assumption. only because the hardware has a serial port does not mean one is going to use in such a way.
my suggestion is to add a `yotta` config option to disable retargeting even if the hardware defines `DEVICE_SERIAL`.

0xc0170, 2016-03-04:
> retarget always init the serial port as soon as DEVICE_SERIAL is set.

If you don't use any stdio in your app (printf(), etc), it should not be initialized. code reference: https://github.com/ARMmbed/mbed-drivers/blob/master/source/retarget.cpp#L177

bearsh, 2016-03-04:
sorry you're right I missed that part. hopefully the compiler is smart enough to optimize away the unused code...

talking about retargeting, what do you think about to integrate the Segger RTT as another possible 'backend', maybe selectable by a yotta config option? (of course this would need a new ticket)

0xc0170, 2016-03-04:
> talking about retargeting, what do you think about to integrate the Segger RTT as another possible 'backend', maybe selectable by a yotta config option? (of course this would need a new ticket)

I am not fully familiar with the implementation, you can open a new ticket for this. If you make it work, let us know.

I am closing this issue.

closed 2016-03-04 by 0xc0170

guest253456, 2018-06-28:
>If you don't use any stdio in your app (printf(), etc), it should not be initialized. code reference: https://github.com/ARMmbed/mbed-drivers/blob/master/source/retarget.cpp#L177

it seems `fault_print_init()` in `mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/mbed_rtx_fault_handler.c` initializes serial and uses it for printing.

0xc0170, 2018-06-28:
@guest253456 This issue is for older codebase that had a bit different retarget (see it points to mbed-drivers repo). 

If there is an issue now , even if it is similar, please file a new issue