ARMmbed/mbed-drivers #186: AnalogIn : Need a new function to return the ADC raw data

bcostm, 2016-04-17:
Today the ADC data is scaled to 16bits. It is necessary in some applications to return the ADC raw data. A new function can be added to the AnalogIn api:
For example:
   unsigned short read_raw() { 
        return analogin_read_raw(&_adc); 
    }

markus-becker-tridonic-com, 2016-04-18:
This would be highly useful, for e.g. CPU internal temperature 12-bit ADCs, where datasheets give you conversion formulas, that assume 12-bit non-scaled raw values as input.

markus-becker-tridonic-com, 2016-04-20:
> Yes, it returns a float and then we'll have to multiply by 0xFFF and convert to a uint16_t... This is why I asked if this method was ok for you...

@mlnx @bcostm I don't think it is a good idea to go this way if the solution is so easy: adc_read should be part of the API.

The workaround involves float calculations which are completely unnecessary and might introduce additional problems.

mlnx, 2016-04-20:
@0xc0170 please review this

0xc0170, 2016-04-20:
It sound sensible.

It should probably return unsigned int value (there are ADC on some chips -  for instance sigma delta with 24bit).

mlnx, 2016-04-29:
Update: under internal discussion at ARM mbed.

anpilog, 2016-06-12:
Hi guys,
Any decision so far?

mlnx, 2016-06-13:
This is unlikely to be fixed soon.

closed 2016-06-13 by mlnx

markus-becker-tridonic-com, 2016-06-14:
Is it because it's non-sense, or because its going to be deprecated soon? If neither, this should be kept open, so that it still can be tracked, or?