ARMmbed/mbed-drivers #188: Support for semihosting

omidontop, 2016-05-25:
It would be nice to allow support for semihosting for targets that have no serial output possibilities. In order to achieve that, one needs to add **--specs=rdimon.specs** to the linker options. But then since the low-level functions defined in rdimon.specs are already defined in retarget.cpp, the program will not compile or work. These functions are:

- _lseek
- _isatty
- _open
- _close

I can see that in device.h file of some mbed-hal-* modules, **DEVICE_SEMIHOST** exists but it doesn't seem to have been used at all. One suggestion that I've tried and worked is to #if !DEVICE_SEMIHOST the definitions of those functions out. The two functions below are also necessary to exclude for semihosting to work correctly:

- _read
- _write

I don't know if I can send a pull request or not.

omidontop, 2016-05-26:
In addition, it is often needed to override the _write for example when one wants to use the SWO channel as the trace output. It would be very nice if the _write definition in retarget.cpp is declared "weak" so that one can define his own _write function.

Kukunin, 2017-01-22:
Spent at least 4 hours trying to enable semihosting in my mbed-based project.
Eventually figured out, that mbed implements low-level functions strictly to redirect them to UART.
So, 👍 for this issue

amq, 2017-04-18:
Just noticed https://github.com/ARMmbed/swo

It's not exactly `printf()` or `tr_debug()`, but it at least allows SWO output for devices without `STDIO_MESSAGES`