ARMmbed/mbed-drivers #194: Mbed-cli and Windows serial driver does not finish installation for many dev boards (critical bug) (updated)

Navis-Raven, 2018-08-29:
All of the bug is described here:
https://github.com/ARMmbed/mbed-cli/issues/733

Screenshot
[image]

Last conclusion: The problem is not a problem of a particular OS (same problem is in windows 7 as well as on windows 8.1), nor a problem of dev board (same problem on both of nRF52-DK and B-L475E-IOT01A), it's a more general problem and critical bug !

0xc0170, 2018-08-29:
Not related to this repository, cli is the correct one to report this one. I'll close this

closed 2018-08-29 by 0xc0170

Navis-Raven, 2018-08-29:
@0xc0170  Ok so the problem is from cli ? but I had the same problem with "Windows serial driver" downloadable here: https://os.mbed.com/docs/latest/tutorials/windows-serial-driver.html

0xc0170, 2018-08-29:
this repository is not for this type of the problem. The name might confuse you as its mbed-drivers. This type of problem is better to report to cli repository or ask on Mbed forums (our support will pick this up from there).

Navis-Raven, 2018-08-29:
Ok